pitronicteam/FOC_ESC-hardware #3: Missing resistor vs reference design.

jaxxzer, 2019-05-06:
https://active-semi.com/wp-content/uploads/PAC5523_Datasheet.pdf

Datasheet Figure 10-2 shows resistor between vcc_in and VHM. It is not present in your design:

https://github.com/AirFleetTeam/FOC_ESC-hardware/blob/master/files/FOC_ESC%20Schematic%20V1.0.pdf

DusKing1, 2019-05-06:
I've checked PAC5523EVK1-RevA-SCH, the R33 between vcc_in and VHM is 0R, and EVK1 can run up to 70V, so I think current schematic is ok.

jaxxzer, 2019-05-06:
@DusKing1 Where did you access the schematic for the evaluation kit? I cannot find it. This is the only document about it that I could find: https://active-semi.com/wp-content/uploads/PAC5523EVK1_UserManual.pdf

closed 2019-05-10 by DusKing1